RussTedrake/manipulation #47: replace scikit-learn nearest neighbor with open3d

RussTedrake, 2020-09-24:
and remove the (unnecessary) dependencies of scipy and scikit-learn 
will wait til pset 2 is fully submitted.

cc @hjsuh94

hjsuh94, 2020-09-24:
Seems like a good idea. I also got the following error while trying to use scikit-learn in Docker: 
`RuntimeError: module compiled against API version 0xc but this version of numpy is 0xb` 

which stopped after I did `pip3 install numpy --upgrade`. 

Strange, but it seems like the version installed by `sudo apt-get install python3-numpy` in `install_prereqs.sh` is not high enough? Would it be good to put a version requirement for `numpy` in `requirements.txt`?

RussTedrake, 2020-09-24:
ugh.  i already did something similar for scipy to make CI happy.  i'm ok if you need to do it.

RussTedrake, 2020-10-04:
@hjsuh94 -- i've made the change in my pose.ipynb.  can you propagate it to the exercises, and let me know?  then we can remove the build dependencies, too.

hjsuh94, 2020-10-04:
Got it! I will make a PR after running some checks like uploading to Gradescope. 

Meanwhile, I think you have a typo in `pose.ipynb`. 
`distance  = np.linalg.norm(point_cloud_A[:,i] - point_cloud_B[:,i])` 

should be 
`distances[i]  = np.linalg.norm(point_cloud_A[:,i] - point_cloud_B[:,indices[i])]`?

RussTedrake, 2020-10-04:
oops.  gulp.  that actually also revealed that i'm not actually using the distances anywhere.  So I've simply removed that part.  thanks for catching it!

PR #59 (2020-10-07): remove scipy and scikit-learn
resolves #47

---
This change is [[image]](https://reviewable.io/reviews/russtedrake/manipulation/59)
commit 87186cd8 (2020-10-07): remove scipy and scikit-learn (#59)

resolves #47
closed 2020-10-07 by RussTedrake