RussTedrake/manipulation #54: docstring conflict between yapf here and drake

RussTedrake, 2020-09-29:
If I copy a file from drake that passes drake's linter (which forced me to put docstrings on a new line), then I get

```
manipulation/jupyter_widgets.py:1 at module level:
        D212: Multi-line docstring summary should start at the first line
manipulation/jupyter_widgets.py:38 in public method `__init__`:
        D212: Multi-line docstring summary should start at the first line
manipulation/jupyter_widgets.py:102 in public method `set_positions`:
        D212: Multi-line docstring summary should start at the first line
manipulation/jupyter_widgets.py:119 in public method `set_joint_positions`:
        D212: Multi-line docstring summary should start at the first line
manipulation/jupyter_widgets.py:152 in public function `MakeJointSlidersThatPublishOnCallback`:
        D212: Multi-line docstring summary should start at the first line
```

@jamiesnape -- can we configure yapf here to either ignore that warning or make it compatible?

jamiesnape, 2020-09-29:
I will double check, but I think we need to swap `D212` and `D213`. We actually linted docstrings long before drake...

jamiesnape, 2020-09-30:
(Actually I am surprised Drake enforced the contrary or anything regarding docstrings. I need to see why that is too.)

PR #55 (2020-10-01): Tweak linter configurations to closer match drake
We will need to make the same changes to underactuated and fix up htmlbook. Closes #54.

---
This change is [[image]](https://reviewable.io/reviews/russtedrake/manipulation/55)
closed 2020-10-01 by RussTedrake