RussTedrake/manipulation #127: Issues with meshcat visualization in course colabs

wf34, 2021-01-30:
I was taking manipulation course self-paced and today I saw the error, which never appeared before:
```
Tunnel 2306fa0eaf41.ngrok.io not found
```
I reproduce this issue in several notebooks and over few different internet options, available to me.
Was the visualization server taken down?

...github may be considered not the best place to raise this issue, so I apologize for that.

ShigekazuIshihara, 2021-01-31:
Dear Prof. Tedrake and colleagues, students,

I am Prof. Shigekazu Ishihara, at the Prosthetics and Orthotics section of Faculty of Rehabilitation, Hiroshima International University. This autumn-winter term, I have utilized your course for the final part of robotics and sensing lecture (for application for robotic artificial limb) at my on-demand lecture in this difficult situation. 
Thank you very very much. 

We (I and students) faced exactly same problem which "wf34" mentioned.
I guess ngrok.io changed some tunneling feature or address.
Please give some clue to ease the problem.

Shigekazu Ishihara, Prof. PhD.
Hiroshima International Univ.

ShigekazuIshihara, 2021-02-01:
we have to make account and set ngrok tunneling?

ShigekazuIshihara, 2021-02-02:
Thank you very much.. Now I see ngrok works well!!

wf34, 2021-02-02:
Thank you so much, the issue is resolved.

closed 2021-02-02 by wf34