RussTedrake/manipulation #130: can I purge open3d dependency now that more point cloud tools are available in drake?

RussTedrake, 2021-02-05:


EricCousineau-TRI, 2022-09-05:
Curious - what's the negative impact of `open3d`? Just that it has a large footprint?

RussTedrake, 2022-09-05:
1) i haven't been able to support it on mac for some time due to an incompatability.
2) converting back and forth between drake point clouds and open3d point clouds is unnecessarily confusing for students.
3) I'm barely using it!

RussTedrake, 2022-09-05:
(plus the large footprint  ;-)

EricCousineau-TRI, 2022-09-05:
Gotcha, that all makes sense, thanks for the explanation!

RussTedrake, 2022-09-06:
I just confirmed, open3d doesn't appear to support python 3.10 yet (which is what Drake uses on mac)
http://www.open3d.org/docs/release/getting_started.html#development-version-pip

```
pip3 install open3d
ERROR: Could not find a version that satisfies the requirement open3d (from versions: none)
ERROR: No matching distribution found for open3d
```

adamconkey, 2022-09-22:
Once this issue is addressed, will a pip install of this package be supported for Python 3.10 on Mac? I've currently been running into the problem you describe where it can't install due to not finding a supported open3d dependency.

RussTedrake, 2022-09-23:
Yes, I believe that should work.  And i've completed converting all but one final pset question; this work should be complete by this weekend.

closed 2022-09-30 by RussTedrake

adamconkey, 2022-10-03:
Thanks for this, I tried the pip install on MacOS today and it works great.

commit 164990cc (2022-09-30): Remove open3d as a dependency

Resolves #130.