RussTedrake/manipulation #137: stochastic_optimization.ipynb from manipulation online couse stopped running

wf34, 2021-02-27:
The version of `stochastic_optimization.ipynb` available in course colabs has stopped installing for me, failing with the error:
```
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
<ipython-input-7-a9ee8c645c2f> in <module>()
      8     #            "setup_manipulation_colab.py")
      9     from setup_manipulation_colab import setup_manipulation
---> 10     setup_manipulation(manipulation_sha='14145ea656da5bdeb7a7ae4fbd7e28b8e464e544', drake_version='0.25.0', drake_build='releases')
     11 
     12 from IPython import get_ipython

2 frames
/content/setup_drake_colab.py in setup_drake(version, build)
    110     # Confirm that we now have pydrake on the path.
    111     assert spec is not None, (
--> 112         "Installation failed.  find_spec('pydrake') returned None.")
    113     assert path in spec.origin, (
    114         "Installation failed.  find_spec is locating pydrake, but not in the "

AssertionError: Installation failed.  find_spec('pydrake') returned None.
```
Could this be resolved?

wf34, 2021-02-27:
looks like it reproduces for other notebooks too
(have tried `policy_gradient.ipynb`)

RussTedrake, 2021-02-27:
This is due to google colab updating (without warning) from python3.6 to python3.7
I have updated the installation script in `intro.ipynb` and the early chapters (if you grab the first cell from that notebook, then these notebooks should work, too).  I'll update the rest very soon!

RussTedrake, 2021-02-27:
oops.  it's the `intro.ipynb` from my underactuated notes that I've updated.  i'll fix this one now, too.

RussTedrake, 2021-02-27:
These notebooks (and most notebooks) should work again now, albeit with a few warnings on installation that will take some more cleanup.  Do let me know if you find any other problems.

closed 2021-02-27 by RussTedrake