RussTedrake/manipulation #149: [deepnote] retrieve or reconsturct the tunneling link url from the notebook

RussTedrake, 2021-08-31:
when the environment has tunneling enabled, I would like to be able to print out the link that users should point their browsers to from inside my code.  how can I reconstruct the proper URL?

I would also need a way to reliably test if the code is running on deepnote or not.  Presumably this would be resolved by whatever method I use to get the URL.

RussTedrake, 2021-08-31:
+@danzvara

RussTedrake, 2021-08-31:
It looks like this should do the trick, assuming that url naming is stable
```
print("https://" + os.environ["DEEPNOTE_PROJECT_ID"] + ".deepnoteproject.com")
```

https://deepnote.com/project/Manipulation-apD-4pzoRo6a69iilJkS9Q/%2Ftesturl.ipynb

(and it looks like I can link to specific notebooks inside the project)

closed 2021-08-31 by RussTedrake

danzvara, 2021-08-31:
Perfect, you found it faster than I. This is the recommended solution