RussTedrake/manipulation #150: [deepnote] can I share a link directly to an entry in the table of contents?

RussTedrake, 2021-08-31:
It would be great to be able to share a link directly to a section of the notebook.  
[image]

danzvara, 2021-09-07:
This should be possible - try doing a right-click on respective section and copy the link, it should point to that section

RussTedrake, 2021-09-07:
Indeed.  Sorry I missed that!

closed 2021-09-07 by RussTedrake

RussTedrake, 2021-09-08:
Unfortunately, I still don't think this will work as I hoped (allowing someone reading the text follows along by launching deepnote at the top, then clicking on the links in html to advance).  To do that, I would need to link to the future _duplicated_ project.