RussTedrake/manipulation #152: [deepnote] Share notebooks with 'execute' permissions

RussTedrake, 2021-09-05:
Posted as a suggestion here:
https://community.deepnote.com/c/suggestions/output-cells-for-anyone-with-the-link-can-execute-this-project?

danzvara, 2021-09-07:
🙏  I replied in the community, but TLDR is that shared outputs are quite central to many collaboration workflows - there should be a simpler solution to get around the tunnel link sharing issue.

RussTedrake, 2021-09-07:
Thanks!  Your answer also made me realize that execute sharing is a potentially awesome, but different feature -- if all users executing the notebook are running on the same provisioned machine, then things get complicated.  Each time the cell with `StartMeshcat` is run, it will try to start a server on the 8080 port.  If that fails, it will go to another ngrok port, etc.

Even if all users point their browser to the deepnote project id url, then the render will get whichever server was successfully launched at the 8080 port, and all of the sliders across different users will interact.

Long story short -- this is good to know.  I'll stick with the "Duplicate" workflow!

closed 2021-09-07 by RussTedrake