RussTedrake/manipulation #191: install nginx in deepnote docker file

RussTedrake, 2022-09-09:


RussTedrake, 2022-09-09:
done.  I now call `RUN /opt/drake/share/drake/setup/deepnote/install_nginx` in the Dockerfile, but of course the service is not started each time the dockerfile is provisioned on Deepnote.  The result is that the first call to StartMeshcat still says "installing nginx", but it at least returns very fast.

I tried adding `RUN systemctl enable nginx` to the dockerfile, but learned the docker does use sysctl / system.d.  Adding `CMD service nginx start` didn't seem to get the job done, either.

cc @jwnimmer-tri, in case you have any better ideas.  But I'm at least happy that it's faster.

jwnimmer-tri, 2022-09-13:
This all sounds fine.

FYI what we do for Drake tutorials is shown here:

https://drake.mit.edu/release_playbook.html#post-release-tutorials-updates

The only slow part is the `apt`, which we bake into the Docker image like so:

```
RUN apt-get -q update && apt-get -q install -y --no-install-recommends nginx-light xvfb && apt-get -q clean
```

We use an `init.ipynb` to do the `install_xvfb`.  I realize now that it could probably do the `install_nginx` as well, which would maybe remove the printout from the actual notebooks for you?

RussTedrake, 2022-09-14:
Thanks.  I thought about `init.ipynb` for this.  I'll go ahead with that.

RussTedrake, 2024-12-28:
I will finish this once https://github.com/RobotLocomotion/drake/issues/22085 happens.

closed 2026-09-07 by RussTedrake