RussTedrake/manipulation #218: math: Hard to see vector norm notation at normal zoom in Firefox?

EricCousineau-TRI, 2022-10-27:
On Firefox 105.0 (Ubuntu 20.04), as of fbc1580:
https://manipulation.csail.mit.edu/pick.html#diff_ik_w_constraints

At normal zoom:
> [image]

At 110% zoom, then it appears:
> [image]

RussTedrake, 2022-10-28:
Thanks for reporting that.  That's frustrating.  It shows ok in Firefox on my mac.  I will look at it on Ubuntu soon. 

I think my main recourse would be to see if upgrading MathJax helps.

RussTedrake, 2026-09-12:
seems aok, and i've upgraded to mathjax 4, so will close this one...

closed 2026-09-12 by RussTedrake