RussTedrake/manipulation #229: failed to load manipulation package even though I did install the package

riochuong, 2022-11-22:
---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
Cell In [4], line 16
      6 #file: package://manipulation/models/clutter.dmd.yaml
      7 model_directives = """
      8 directives:
      9 - add_directives:
   (...)
     13     file: package://drake/examples/manipulation_station/models/061_foam_brick.sdf
     14 """
     15 station = builder.AddSystem(
---> 16     MakeManipulationStation(model_directives=model_directives))
     17 plant = station.GetSubsystemByName("plant")
     18 plant.SetDefaultFreeBodyPose(plant.GetBodyByName("base_link"), X_O['initial'])

File ~/drake/lib/python3.10/site-packages/manipulation/scenarios.py:529, in MakeManipulationStation(model_directives, filename, time_step, iiwa_prefix, wsg_prefix, camera_prefix, package_xmls)
    527 if model_directives:
    528     directives = LoadModelDirectivesFromString(model_directives)
--> 529     ProcessModelDirectives(directives, parser)
    530 if filename:
    531     parser.AddAllModelsFromFile(filename)

RuntimeError: error: URI 'package://manipulation/models/clutter.dmd.yaml' refers to unknown package 'manipulation'

riochuong, 2022-11-22:
looks like the models folder missing the package.xml file if installing directly from pypi ?? :)

RussTedrake, 2022-11-23:
You are absolutely right.  🤦‍♂️   Thanks for catching it.  I'm rolling a new pip wheel now.

closed 2022-11-23 by RussTedrake

MichaelJFishman, 2023-05-20:
I believe this is broken again. I resolved it locally by adding 

```
include manipulation/models/*.yaml manipulation/models/*.xml
```

to `MANIFEST.in` and then `pip install .`.

I'm guessing we also want to add everything else in `manipulation/models` to `Manifest.in`, but these are the the only parts I've needed so far for the [pick](https://github.com/RussTedrake/manipulation/blob/master/pick/pick.ipynb) and [kinematic tree](https://github.com/RussTedrake/manipulation/blob/master/pick/kinematic_tree.ipynb) notebooks.

RussTedrake, 2023-05-30:
I've set up pip on my CI to make sure i don't have regressions on this.  But it succeeds on master without any changes.

And building the pip wheels says that those files are included....
```
adding 'manipulation/BUILD.bazel'
adding 'manipulation/__init__.py'
adding 'manipulation/clutter.py'
adding 'manipulation/drake_gym.py'
adding 'manipulation/icp.py'
adding 'manipulation/meshcat_cpp_utils.py'
adding 'manipulation/meshcat_utils.py'
adding 'manipulation/mustard_depth_camera_example.py'
adding 'manipulation/pick.py'
adding 'manipulation/scenarios.py'
adding 'manipulation/utils.py'
adding 'manipulation/envs/BUILD.bazel'
adding 'manipulation/envs/box_flipup.py'
adding 'manipulation/exercises/BUILD.bazel'
adding 'manipulation/exercises/grader.py'
adding 'manipulation/exercises/clutter/BUILD.bazel'
adding 'manipulation/exercises/clutter/normal_solution.npy'
adding 'manipulation/exercises/clutter/test_analytic_grasp.py'
adding 'manipulation/exercises/clutter/test_grasp_candidate.py'
adding 'manipulation/exercises/clutter/test_normal.py'
adding 'manipulation/exercises/clutter/test_simulation_tuning.py'
adding 'manipulation/exercises/deep_perception/BUILD.bazel'
adding 'manipulation/exercises/deep_perception/test_contrastive.py'
adding 'manipulation/exercises/force/BUILD.bazel'
adding 'manipulation/exercises/force/test_hybrid.py'
adding 'manipulation/exercises/pick/BUILD.bazel'
adding 'manipulation/exercises/pick/plot_planar_manipulator.py'
adding 'manipulation/exercises/pick/test_differential_ik.py'
adding 'manipulation/exercises/pick/test_planar_manipulator.py'
adding 'manipulation/exercises/pick/test_rigid_transforms.py'
adding 'manipulation/exercises/pick/test_robot_painter.py'
adding 'manipulation/exercises/pick/test_simple_qp.py'
adding 'manipulation/exercises/pose/BUILD.bazel'
adding 'manipulation/exercises/pose/pcl_filtered.npy'
adding 'manipulation/exercises/pose/test_icp.py'
adding 'manipulation/exercises/pose/test_pose_estimation.py'
adding 'manipulation/exercises/pose/test_ransac.py'
adding 'manipulation/exercises/rl/BUILD.bazel'
adding 'manipulation/exercises/rl/test_stochastic_optimization.py'
adding 'manipulation/exercises/rl/test_vpg.py'
adding 'manipulation/exercises/robot/BUILD.bazel'
adding 'manipulation/exercises/robot/test_direct_joint_control.py'
adding 'manipulation/exercises/robot/test_manipulation_io.py'
adding 'manipulation/exercises/robot/test_reflected_inertia.py'
adding 'manipulation/exercises/robot/test_survey.py'
adding 'manipulation/exercises/segmentation/BUILD.bazel'
adding 'manipulation/exercises/segmentation/test_mask.py'
adding 'manipulation/exercises/segmentation/test_segmentation_and_grasp.py'
adding 'manipulation/exercises/trajectories/BUILD.bazel'
adding 'manipulation/exercises/trajectories/test_door_opening.py'
adding 'manipulation/exercises/trajectories/test_rrt_planning.py'
adding 'manipulation/exercises/trajectories/test_taskspace_iris.py'
adding 'manipulation/exercises/trajectories/rrt_planner/BUILD.bazel'
adding 'manipulation/exercises/trajectories/rrt_planner/__init__.py'
adding 'manipulation/exercises/trajectories/rrt_planner/geometry.py'
adding 'manipulation/exercises/trajectories/rrt_planner/iiwa_rrt_problem.py'
adding 'manipulation/exercises/trajectories/rrt_planner/robot.py'
adding 'manipulation/exercises/trajectories/rrt_planner/rrt_planning.py'
adding 'manipulation/models/061_foam_brick_w_visual_contact_spheres.sdf'
adding 'manipulation/models/BUILD.bazel'
adding 'manipulation/models/camera_box.sdf'
adding 'manipulation/models/clutter.dmd.yaml'
adding 'manipulation/models/clutter_mustard.dmd.yaml'
adding 'manipulation/models/clutter_planning.dmd.yaml'
adding 'manipulation/models/clutter_w_cameras.dmd.yaml'
adding 'manipulation/models/double_pendulum.urdf'
adding 'manipulation/models/floor.sdf'
adding 'manipulation/models/iiwa_and_wsg.dmd.yaml'
adding 'manipulation/models/manipulation_station_with_cupboard.dmd.yaml'
adding 'manipulation/models/mustard_w_cameras.dmd.yaml'
adding 'manipulation/models/package.xml'
adding 'manipulation/models/planar_bin.sdf'
adding 'manipulation/models/planar_foam_brick_collision_as_visual.sdf'
adding 'manipulation/models/planar_manipulation_station.dmd.yaml'
adding 'manipulation/models/rubiks_cube.sdf'
adding 'manipulation/models/rubiks_cube_2_by_2.sdf'
adding 'manipulation/models/schunk_wsg_50_welded_fingers.dmd.yaml'
adding 'manipulation/models/schunk_wsg_50_welded_fingers.sdf'
adding 'manipulation/models/schunk_wsg_50_welded_fingers_sphere.sdf'
adding 'manipulation/models/segmentation_and_grasp_scene.dmd.yaml'
adding 'manipulation/models/shelves.sdf'
adding 'manipulation/models/shelves_w_ellipsoid_collision.sdf'
adding 'manipulation/models/two_bins.dmd.yaml'
adding 'manipulation/models/two_bins.sdf'
adding 'manipulation/models/two_bins_w_cameras.dmd.yaml'
adding 'manipulation/models/two_bins_w_cameras.sdf'
adding 'manipulation/models/two_link_iiwa14.urdf'
adding 'manipulation/models/bunny/BUILD.bazel'
adding 'manipulation/models/bunny/README'
adding 'manipulation/models/bunny/bun_zipper_res2.ply'
adding 'manipulation/models/bunny/bunny.npy'
adding 'rl/train_boxflipup.py'
adding 'manipulation-2023.5.29.dist-info/LICENSE.TXT'
adding 'manipulation-2023.5.29.dist-info/METADATA'
adding 'manipulation-2023.5.29.dist-info/WHEEL'
adding 'manipulation-2023.5.29.dist-info/top_level.txt'
adding 'manipulation-2023.5.29.dist-info/RECORD'
```

so I'm not sure what happened in your case?

MichaelJFishman, 2023-06-04:
I'm not sure either. Maybe it's related to my using conda? Drake has a [warning](https://drake.mit.edu/pip.html#stable-releases) that anaconda may cause issues:

> Drake does not support the Python environment supplied by Anaconda. Before installing or using Drake, please conda deactivate (repeatedly, until even the conda base environment has been deactivated) such that none of the paths reported which -a python python3 pip pip3 refer to conda. Note that Miniconda seems to work fine; it’s only Anaconda that has caused problems for some users.

commit 76e24aa3 (2022-11-23): Add *.xml to manifest

Resolves #229.