RussTedrake/manipulation #232: Multiple instantiations of `MultiBodyPlant` and `iiwa`.

Kzernobog, 2023-04-01:
I am having trouble understanding why the iiwa model is being  loaded [here](https://github.com/RussTedrake/manipulation/blob/396fe7c40b5881a58bbb5b8f083b70c1c11bd94f/manipulation/scenarios.py#L736) again. Wasn't the iiwa robot being loaded via the the model directives in the function [here](https://github.com/RussTedrake/manipulation/blob/396fe7c40b5881a58bbb5b8f083b70c1c11bd94f/manipulation/scenarios.py#L684)? 

Another question I had was what was the reason you have maintained two different plants for the main system and the controller when setting up the manipulation station. Couldn't we add all the elements to the same instance of the Multibody plant? Am I missing something here?

Thank you for your work and instruction, it has been very helpful.

RussTedrake, 2023-04-02:
Having two different models is the most common workflow, and I think appropriate.  The first model (called `plant`) is the model used by the physics engine; it includes both the robot and the objects in the world.  The second model (called here `controller_plant`) is the robot's "internal model" used for control; it only includes the robot itself (and in fact has a welded version of the gripper).  This corresponds to the model that the iiwa control station has in its software stack; which really does have only the dynamics of the arm + a rigid "tool" inertia.

closed 2023-04-02 by RussTedrake

Kzernobog, 2023-04-03:
Ah I see, I misunderstood the `plant` to be the physics engine. Does this mean that if you load `n` robots into the scene, your code needs to always have `n+1` instantiations of `MultiBodyPlant` - in the case of maintaining a controller for each robot?

RussTedrake, 2023-04-03:
In this code, the `plant` *is* the physics engine, and the `controller_plant` is the internal model.  You can model all sorts of combinations of things, but yes, I would say if you have `n` robots that all have their own internal models, then you would likely end up with `n+1` plants.