RussTedrake/manipulation #233: from pydrake.geometry import RenderLabel

buoyancy99, 2023-04-27:
I am installing the repo and drake nightly with
```
pip install -U https://drake-packages.csail.mit.edu/drake/nightly/drake-latest-cp310-cp310-manylinux_2_31_x86_64.whl
pip install git+https://github.com/RussTedrake/manipulation@master
```

Then if we do `from manipulation import running_as_notebook`, an error would rise with `ImportError: cannot import name 'RenderLabel' from 'pydrake.geometry' `

I am wondering if we shall change this [line](https://github.com/RussTedrake/manipulation/blob/01d2a4bf2b08bb33aeb407bf1258ea396f2b7a50/manipulation/utils.py#L9) to ` from pydrake.geometry.render import RenderLabel`

buoyancy99, 2023-04-27:
PR [here](https://github.com/RussTedrake/manipulation/pull/234)

RussTedrake, 2023-04-27:
I assume you realized you needed to update drake?

closed 2023-04-27 by RussTedrake