RussTedrake/manipulation #237: ModuleNotFoundError in Mask R-CNN for Bin Picking

adityaanirudhk, 2023-06-29:
Hi. Im trying to run the code provided in example 9.2 on Google Colab and I'm receiving a ModuleNotFoundError. I have tried to find the root cause for this problem but couldn't find any. It would be helpful if you could suggest another way to run the code.
[image]

cohnt, 2023-06-30:
Make sure you `pip install manipulation`, otherwise the package won't be available. To run everything in this notebook on colab, I needed `!pip install drake manipulation pyvirtualdisplay xvfbwrapper` and `!apt install xvfb` at the top. You also may find that Deepnote supports the textbook examples a bit more easily than Colab.

adityaanirudhk, 2023-07-01:
Thank you. I'll make sure I check that out.

RussTedrake, 2023-07-01:
Thanks @cohnt !

closed 2023-07-01 by RussTedrake