RussTedrake/manipulation #300: update or remove trimesh dependency

RussTedrake, 2024-03-25:
I've pegged the version of `trimesh` to be `<4.2.0`, because the newest version of trimesh added a dependency on `openctm` (technically only as "recommends"), but `openctm` does not seem to support mac arm.

I've had trouble getting poetry to export optional dependencies reliably. An alternative possible solution would be to figure out how to install trimesh without these dependencies.

But trimesh appears to be regularly breaking things. I think it's not a good dependency to have to keep around, and would like to consider purging it.

nepfaff, 2025-01-07:
This seems fixed now that we have optional dependencies?

RussTedrake, 2025-01-07:
Having trimesh at all has, at times, been a maintenance burden, since they make changes like the one listed above. I don't think the 'extras' improvement solved that?

nepfaff, 2025-01-22:
From the trimesh maintainer:

>Thanks for the PR! Just out of curiosity, how are you managing your trimesh dependency? There are multiple "helper" extras, none of which are required, and depending on anything other than trimesh[easy] is unlikely to work on all platforms.
For upstream projects, it's a lot easier to manage the poetry/UV/requirements lock files with exactly the soft dependencies that a project actually uses, i.e. trimesh, scipy, xatlas rather than using an extra, since just trimesh will only pull in numpy.

This might be the best way to go. Trimesh just depending on numpy is not bad and we can't benefit from any non-Mac compatible optional dependencies anyways. I noticed that we already have `vhacdx` as a separate dependency.

RussTedrake, 2025-01-22:
Sounds good to me!

nepfaff, 2025-01-22:
okay, will link as completed with that PR

closed 2025-01-23 by nepfaff