RussTedrake/manipulation #308: Deepnote python package manipulation version is different than the latest? Seeing errors.

jtcook2, 2024-04-28:
First, thanks for all the effort put into the course and making the resources accessible.

I'm trying to type/ implement the deepnote examples in the lectures locally on my machine as I follow along to make sure my understanding is clear and to learn the Drake library simultaneously.

I installed both drake and manipulation(https://pypi.org/project/manipulation/) in a python environment via the directions here: https://drake.mit.edu/pip.html#stable-releases. 

However, when I try to implement the Robot Arms example (https://deepnote.com/workspace/Manipulation-ac8201a1-470a-4c77-afd0-2cc45bc229ff/project/8f86172b-b597-4ceb-9bad-92d11ac7a6cc/notebook/inspector-7b7d6cf641e04ac5a70c3150c12da475) I get errors. I looked in my virtual environment and it doesn't seem the filesystem structure of my manipulation package matches what's on Deepnote (picture attached)
[image]

Are there specific versions of Drake and manipulation that should be used together to recreate the Deepnote experiences locally? Thanks for taking the time to read my question!

RussTedrake, 2024-04-28:
It's true that Drake moved some of the packages recently.  `pip install manipulation` (with the latest release, https://pypi.org/project/manipulation/#history) should always get you pip version and a matching Drake version.  What happened here is that I haven't pushed the updated notebooks to Deepnote recently, since I'm not teaching manipulation this term.  Thanks for letting me know.  I've pushed them now.

Downloading the notebook in this repository should also always work with the latest manipulation pip wheels.  (I don't have deepnote under CI directly, but do have the pip workflow on the main branch of this repo under test).

closed 2024-04-28 by RussTedrake

jtcook2, 2024-04-29:
Thanks for the quick feedback! The [robot arm example](https://deepnote.com/workspace/Manipulation-ac8201a1-470a-4c77-afd0-2cc45bc229ff/project/8f86172b-b597-4ceb-9bad-92d11ac7a6cc/notebook/inspector-7b7d6cf641e04ac5a70c3150c12da475) now works locally on my machine.

Just in case it's useful to others: I did need to install and use Python3.11 for my virtual environment. Otherwise I got package incompatibility errors. The  [webpage](https://pypi.org/project/manipulation/2024.4.22/#description) seems to indicate Python3.10 works. Also you need to manually add ipython to your virtual environment or the manipulation package errors on import.

[This link](https://apple.stackexchange.com/questions/450303/how-to-make-python-3-11-my-default-python3-with-brew) was helpful for Python via home-brew. I thought I would include it since pydrake installation instructions emphasize using homebrew on macOS.

RussTedrake, 2024-05-05:
Thanks @jtcook2 .   Just for clarity on the python versions... they are platform dependent, and chase primarily Drake's support matrix:  https://drake.mit.edu/installation.html