RussTedrake/manipulation #322: Adding frozen children doesn't support `add_frame` directives

siddancha, 2024-07-06:
## Current logic

When `add_frozen_child_instances=True`, the current logic in `_PopulatePlantOrDiagram()` searches over a tree whose edges are defined only by weld directives to find child instances to add (see `_FindChildren()`).

[[Permalink]](https://github.com/RussTedrake/manipulation/blob/ee87f28c8b4b0b2f3b9f76ee73d0c9b5d7c9c6c9/manipulation/station.py#L340-L349):

```python
tree = dict()
children = set()
for d in flattened_directives:
    if d.add_weld:
        parent = ScopedName.Parse(d.add_weld.parent).get_namespace()
        child = ScopedName.Parse(d.add_weld.child).get_namespace()
        if parent not in tree:
            tree[parent] = {child}
        else:
            tree[parent].add(child)
```

Furthermore, only two types of directives are retained to be added to the controller plant -- `add_model` and `add_weld`:

[[Permalink]](https://github.com/RussTedrake/manipulation/blob/ee87f28c8b4b0b2f3b9f76ee73d0c9b5d7c9c6c9/manipulation/station.py#L431-L447):

```python
directives = []
for d in flattened_directives:
    if d.add_model and (d.add_model.name in all_model_instances):
        directives.append(d)
    if (
        d.add_weld
        and (
            ScopedName.Parse(d.add_weld.child).get_namespace()
            in all_model_instances
        )
        and (
            d.add_weld.parent == "world"
            or ScopedName.Parse(d.add_weld.parent).get_namespace()
            in all_model_instances
        )
    ):
        directives.append(d)
```

## The problem

This logic is problematic when the scenario file contains `add_frame` directives. For example, a common workflow is to add a custom frame relative to a body frame of one robot, and weld the custom frame to a body frame of another robot. In this case, the controller plant throws an error because it cannot find the declaration of the custom frame.

Furthermore, `_FindChildren()` will not find children that are separated by at least one `add_frame` directive.

## Proposed solution

We should treat `add_frame` directives as edges just like `add_weld` in the search tree. This will detect any robot instance separated by a sequence of `add_frame` and `add_weld` directives as a child.

Secondly, we should add the relevant `add_frame` directives into the controller plant.

## Temporary workaround

I'm temporarily working around this issue by avoiding `add_frame` directives in my scenario file, and directly welding body frames instead.

RussTedrake, 2024-07-07:
I agree that this is an issue. Good to have it on the queue, and happy to take contributions if you find it easier to resolve it.

siddancha, 2024-07-09:
> I agree that this is an issue. Good to have it on the queue, and happy to take contributions if you find it easier to resolve it.

@RussTedrake Added PR #323 fixing this issue.

PR #323 (2024-07-10): Support `add_frame` directives when adding frozen children
Fixes #322 

## Solution
- [x] Implements a new `DirectivesTree` class in `manipulation/directives_tree.py`. This class creates a tree where:
  - **Nodes:** are defined by _(1)_ model instances and _(2)_ added frames.
  - **Edges:** are defined by _(1)_ `add_weld` and _(2)_ `add_frame` directives.
- [x] The tree implements two recursive search functions:
  - Finding paths from the root ("world" frame) to a given set of robot instances. The use case is to find the minimal set of directives that supports the given robot instances.
  - Fi [... cut]
closed 2024-07-10 by RussTedrake