RussTedrake/manipulation #392: Deepnote's Init notebook doesn't always run on startup

RussTedrake, 2024-12-28:
First observed with @thatprogrammer1 and @nepfaff during a [slack discussion](https://mit-6-4210-2-f24.slack.com/archives/C07GF3QH2CT/p1726177704808099?thread_ts=1726155034.460999&cid=C07GF3QH2CT).

I think it will probably require a discussion with the deepnote support team to resolve.

nepfaff, 2025-01-02:
This is already being processed by the deepnote support team. They have access to one of our projects that demonstrates this issue.

[image]

The contact person is David Hughes ([email](david.hughes@deepnote-7619ccb5d2aa.intercom-mail.com)).

RussTedrake, 2025-02-11:
That's correct.  But that was due to a different issue: #435.

RussTedrake, 2025-09-03:
Some progress on this.  I currently believe that Init will run at startup iff the Init notebook has a lightning bolt next to it in the GUI:

[image]

Some projects did not have that, e.g.:

[image]
I suspect that the reason for this is that Init was created using my deepnote API instead of from the GUI. 

The easy, but annoying, fix seems to be to do the following in the web gui:
- copy the contents of the Init notebook into your copy buffer
- delete the Init notebook
- create a new notebook named Init, verify that it gets a lightning bolt
- paste the old contents into the first cell

That has resolved it each time for me.