RussTedrake/manipulation #395: fix convoluted workflow for PRing changes that require new pip wheels

RussTedrake, 2024-12-29:
- We currently have some CI (in the noncritical.yaml) which test the current repo against the deployed pip wheels).
- We also have a `check_deepnote_requirements` test in the repo that will fail (in the main ci) if the `version` in `pyproject.toml` doesn't match the pip wheels deployed on deepnote.

I should be able to fix this so that local changes can pass the tests, and that CI can warn if a change requires a new pip wheel to be deployed (both in PR and in nightlies).

nepfaff, 2025-01-02:
I have been using the "current alternative" approach so far just before merging to avoid pushing a broken branch. I do agree that this isn't great but I'm not a fan of the first approach either.

An easy workaround for now would be to move the `check_deepnote_requirements` into a separate CI, preventing it from hiding other test failures. I can quickly do that if you agree.
I'm guessing that you mean something like this when saying "that CI can warn if a change requires a new pip wheel to be deployed"? From a quick look, there are no great "warning" type failures that have good visibility.

RussTedrake, 2025-01-02:
> move the check_deepnote_requirements into a separate CI

Yes, I think this is a strict improvement. But the way that I would do this would be to disable that test via an exclusion (e.g. a bazel test tag) in the instances of CI that shouldn't have to check it.  I would still want my `bazel test //...` to check it locally.

closed 2025-01-03 by RussTedrake

nepfaff, 2025-01-04:
The conditional execution solution is great! I didn't know this was an option

commit 3832dfc9 (2025-01-03): Makes it possible to mark a PR as 'requires new pip wheels' (#398)

and thereby disable the two CI actions that would have failed.

Resolves #395.