RussTedrake/manipulation #435: Deepnote notebooks machine failing to start - Deepnote Docker image requirement changes

stevegra, 2025-01-23:
The Manipulation notebooks on Deepnote that I have tried recently are not running right now. After multiple exchanges with Deepnote support (also Dave Hughes), the gist of the issue is that Deepnote added the requirement to have curl in the Docker images, and your public Docker images for the Manipulation notebooks only have bash. So the cached Docker images with all the Manipulation (and presumably the Underactuated) notebooks for your courses won't start on a Deepnote machine.

The temporary fix from Deepnote support and engineers is to add the following lines to the Dockerfile for each notebook (so they appear as follows), rebuild the Docker image, then start the machine.

FROM russtedrake/underactuated:e849980 
RUN apt-get update && apt-get install -y curl python3.10-venv 
RUN apt-get remove -y python3-notebook

I have tried this and it works for the notebooks that I have been using.
I think this unfortunately means that all of your publicly available Docker images that the Dockerfiles in your Deepnote notebooks will have to be updated with these lines or something similar, and the cached Docker images in each of the notebooks for duplication will have to be rebuilt in order for your Deepnote teaching materials to work.

stevegra, 2025-01-23:
The ticket number that I had open with Dave Hughes at Deepnote support was #6037

RussTedrake, 2025-01-25:
All set now.  Thank you very much!

closed 2025-01-25 by RussTedrake