RussTedrake/manipulation #441: Get textures working in scene_synthesizer example

RussTedrake, 2025-02-16:
I've opened https://github.com/NVlabs/scene_synthesizer/issues/10 to ask about it, but currently understand that the textures only work when exporting to USD.

A possible next step is to see how close Drake's fledgling USD parser is to parsing the generated scenes. (I tried a quick experiment on that, but even after building Drake using `WITH_USD=on`, calling the USD parser from installed python immediately throws an error about not being able to find runfiles.

RussTedrake, 2025-02-17:
I could try [usd2sdf from this repo](https://github.com/gazebosim/gz-usd).

rcory, 2025-02-21:
I wanted to see the textured kitchen, so I went ahead and installed Isaac Sim on my laptop. I then went ahead and ran [the materials example](https://scene-synthesizer.github.io/examples/add_mdl_materials.html) from scene synthesizer, which ends up spitting out a USD (which in theory contains textures). 

When attempting to import the resulting USD into Isaac Sim, it immediately throws an error with "Failed to open stage kit.usd. Please check the console for error." Checking the console, however, doesn't provide any useful info (basically says the same message).

I then downloaded a USD asset pack from [here](https://docs.omniverse.nvidia.com/usd/latest/usd_content_samples/downloadable_packs.html#usd-assets) ("Showcases Scene Asset Pack"), and imported the `IsaacWarehouse.usd`  and it rendered beautifully in Isaac Sim (textures and geometry). So I know my installation is good.

Perhaps scene synthesizer requires a different version of IsaacSim...

As an alternative, I also installed [usdview](https://docs.omniverse.nvidia.com/usd/latest/usdview/quickstart.html), and tried opening up the kitchen USD there. usdview was able to import the USD and visualize the kitchen, but it had no textures applied.

Loooking at the usdview console, it generated a bunch of these errors:
```
Warning (secondary thread): in BindingsAtPrim at line 669 of /builds/omniverse/usd-ci/USD/pxr/usd/usdShade/materialBindingAPI.cpp -- Found material bindings on prim at path (/world/countertop_corner/geometry_0) but MaterialBindingAPI is not applied on the prim
```

I ran across this (older) [related post](https://projects.blender.org/blender/blender/issues/107531), which seems to indicate that newer versions of USD have checks on material bindings that older versions did not.

As a third alternative, I also tried to [import the USD in Blender](https://docs.blender.org/manual/en/latest/files/import_export/usd.html), but again resulted in the same behavior as `usdview`, i.e., I was able to visualize the kitchen geometry, but no textures. Blender does have a glTF exporter, so my hope was that if I were to import the USD into Blender (with textures) I would be able to export it to glTF, but no luck there.

Anyway, that was my foray into textures today. For now, I'm going to focus on the models and worry about the textures later.

RussTedrake, 2025-02-21:
ftr -- i noticed that the example code puts the output file in /tmp/kit.usd, but the textures get put into the local directory (not in the /tmp directory. Specifying the `texture_dir` directory caused it throw an error.

I was actually working a bit today on seeing how far we can get using Drake's USD parser on the file that gets spit out. I'll play with that a little more, too.

RussTedrake, 2025-02-21:
Ok... a little progress.  After resolving some bugs in the current usd parser implementation, I'm able to load the geometry directly into Drake from the exported USD. 
[image]

I still need to implement the joint parsing, and the textures, but it will be really good to have this.

RussTedrake, 2025-02-22:
Making a few relevant USD links here for myself:
- https://docs.omniverse.nvidia.com/usd/latest/usdview/quickstart.html
- https://openusd.org/dev/api/class_usd_geom_cube.html
- schemas: https://github.com/PixarAnimationStudios/OpenUSD/blob/release/pxr/usd/usdGeom/schema.usda
- usdcat https://openusd.org/release/toolset.html#usdcat

rcory, 2025-02-24:
> but the textures get put into the local directory

Correct. In the tests above, I also tried manually moving everything into the same directory, with no luck. Also, in my testing, the `textures` directory that gets saved in the local directory has a single `.png` file, with what seems like a mostly black image containing a single grey pixel in the middle. So there's something obviously wrong here.

> I was actually working a bit today on seeing how far we can get using Drake's USD parser on the file that gets spit out

This would be great!

rcory, 2025-03-04:
This question isn't related to textures, but related to USD parsing. Is there a place for USD to define hydroelastic contact parameters? Or, is the expectation that these would be declared outside of the USD file?

RussTedrake, 2025-03-05:
> Is there a place for USD to define hydroelastic contact parameters?

We would have to add hydro support to the USD parser, too (or just enable hydro by default).

Did you see the response in https://github.com/NVlabs/scene_synthesizer/issues/10 ?  He kindly offered a way to make some textures (perhaps not quite as nice) appear in the urdf version.  We could use that to start...

nepfaff, 2025-07-12:
> I could try [usd2sdf from this repo](https://github.com/gazebosim/gz-usd).

I tried this but it replaces all PBR materials with black materials

nepfaff, 2025-07-14:
Making progress with a custom pipeline for converting USD files into Drake SDF files (using a mix of Blender and `usd2sdf`). Not yet great and still messy, but it should get there soon.

Example for a kitchen from the `ArtVIP` dataset that had textures stored in the USD files.
[image]