RussTedrake/manipulation #465: upgrade docker image and test on deepnote

RussTedrake, 2025-08-27:
Perhaps this should happen just after #462.  

I just recently updated bazel and poetry and a number of dependencies in order to work with the latest versions of drake (including python 3.13 support on mac).  I have not yet tested on deepnote. 

Deepnote uses the dockerfile image from this repo as a starting point, then installs the latest pip wheels. I think the first step is:
- [ ] to get the dockerfile workflow working again -- i had to comment out the github actions workflow in main.yml to get things working.  note that we might consider upgrading the docker from jammy => noble; that might depend on deepnote.
- [ ] retest the manipulation notebooks on deepnote.  we should make sure the notebooks for the first week or two all work. but i'm most worried about the ones that use torch, because I had previously pinned torch < 2.4 for deepnote, and had to unpin that to make python 3.13 support happen. Hopefully deepnote has moved forward on this, too.

@bernhardpg -- i'm hoping that you or one of the fall TAs can help with this (or even take it). it will be important for someone to understand this workflow.

RussTedrake, 2025-08-30:
I've gone ahead and updated the dockerfile to use noble. (I did not push this change to deepnote yet, so that you can test one step at a time).  @bernhardpg -- hoping you can take it from here.

nepfaff, 2025-08-31:
@bernhardpg tested the first 2 weeks on deepnote with the new docker file and they work. I updated deepnote with the new docker

nepfaff, 2025-08-31:
Though the docker CI seems to be failing with it not finding the new letter module. I tested the pypi wheel and that module does exist. Maybe some bad caching?

RussTedrake, 2025-09-01:
Thank you both. Did you manually change the docker sha in deepnote, or did you actually run `publish_to_deepnote` (e.g. via https://github.com/RussTedrake/manipulation/blob/master/book/Deepnote.sh) to update all of the deepnote files?

RussTedrake-LBM, 2025-09-02:
Great. Thanks!  I think the last step would be to test some notebook that requires torch. If that seems to work, then we can close this issue.