RussTedrake/manipulation #502: CI no longer runs on pushes to main

nepfaff, 2025-09-10:
The CI no longer runs if we push to `master` directly. See https://github.com/RussTedrake/manipulation/commit/ebcda35c11b8f429c9e8889b2f176e6cf6953823 for an example.

I see two options:

1. Change the PR to always run on pushes to `master`
2. Make `master` a hard protected brunch, forcing even repo admins to create PRs

Option 2 is cleaner. Option 1 is what we have had so far and results in slightly faster development.
@RussTedrake, any preferences?

RussTedrake, 2025-09-10:
This is presumable due to #495 .  @bernhardpg

bernhardpg, 2025-11-12:
FYI @RussTedrake this is now resolved (fixed in #561 ). I pushed [a commit](https://github.com/RussTedrake/manipulation/commit/bb10ec83ea7a3086e0c8b04956e23f3f09456fcb) with some minimal changes to master to test this, and indeed tests are running as desired. Closing this issue.